nordicsemi/Flutter-nRF-Connect-Device-Manager #9: Example app doesn't do anything

reinhardbluelab, 2023-04-24:
Code is commented out

euclidk, 2023-07-06:
If I uncomment the code, it looks like there's a conflict between the version of rxdart that mcumgr_flutter requires and the one that flutter_blue requires.

See:
[example] flutter pub get
Resolving dependencies...
Because every version of mcumgr_flutter from path depends on rxdart ^0.27.7 and flutter_blue >=0.8.0 depends on rxdart ^0.26.0, mcumgr_flutter from path is incompatible with flutter_blue >=0.8.0.
So, because mcumgr_flutter_example depends on both flutter_blue ^0.8.0 and mcumgr_flutter from path, version solving failed.
exit code 1

flutter_blue is no longer being actively maintained but there is an active fork called flutter_blue_plus. I created a fork that uses flutter_blue_plus for the example app and started a pull request here:
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/pull/31

NickKibish, 2024-01-04:
I added an example app. 
Now it's only for Thingy:53. Maybe add something for other boards as well.

closed 2024-01-04 by NickKibish