nordicsemi/Flutter-nRF-Connect-Device-Manager #25: Unable to update firmware - Wrong magic number /

AvivZem, 2023-05-29:
Running on Andrdoid,
When i attempt to update my device firmware i get the following error:

E/DartMessenger: Uncaught exception in binary message listener
    **Wrong magic number (found 0x20010000, expected 0x96F3B83D or 0x96F3B83C)**
        at io.runtime.mcumgr.image.McuMgrImageHeader.fromBytes(McuMgrImageHeader.java:66)
        at io.runtime.mcumgr.image.McuMgrImageHeader.fromBytes(McuMgrImageHeader.java:54)
        at io.runtime.mcumgr.image.McuMgrImage.fromBytes(McuMgrImage.java:89)
        at io.runtime.mcumgr.dfu.FirmwareUpgradeManager.start(FirmwareUpgradeManager.java:436)
        at no.nordicsemi.android.mcumgr_flutter.UpdateManager.start(UpdateManager.kt:82)
        at no.nordicsemi.android.mcumgr_flutter.McumgrFlutterPlugin.update(McumgrFlutterPlugin.kt:151)
        at no.nordicsemi.android.mcumgr_flutter.McumgrFlutterPlugin.onMethodCall(McumgrFlutterPlugin.kt:70)
        at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
        at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:296)
        at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:320)
        at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:233)
        at android.app.ActivityThread.main(ActivityThread.java:8010)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:631)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:978)

Also, i get the following error as well (right after the first one):

Unhandled Exception: Unhandled error MissingPluginException(No implementation found for method update on channel mcumgr_flutter/method_channel)

i used the git dependency as follows:
`mcumgr_flutter:
    git:
      url: https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager.git
      ref: master`

The piece of code i run:
`Future<FirmwareUpdateManager> _unpackData(
      FileSystemEntity file, String deviceId) async {
    final uuid = Uuid().v4();
    final tmpDir = await getTemporaryDirectory();
    final dirPath = path.join(tmpDir.path, uuid);
    final dir = Directory(dirPath);
    File zipFile = File(file.path);
    await ZipFile.extractToDirectory(zipFile: zipFile, destinationDir: dir);
    final files = await dir.listSync();
    final binFile = files.firstWhere(
        (file) => file is File && file.path.endsWith('.bin'),
        orElse: () => null);
    if (binFile == null) {
      throw ("no bin file found in ZIP");
    }
    final fileContent = await File(binFile.path).readAsBytes();
    fwScheme.add(Tuple2(1, fileContent));
    final updateManager =
        await FirmwareUpdateManagerFactory().getUpdateManager(deviceId);
    updateManager.setup();
    await updateManager.update(fwScheme);
    return updateManager;
  }`

The error is when i run the line:

`await updateManager.update(fwScheme);`

The data (Tuple<int, List<int>>) i attempt to send to the device (only 1 Tuple):
I can only show the first numbers (file too long):

[[1, [0, 0, 1, 32, 117, 147, 1, 0, 93, 147, 1, 0, 53, 189, 1, 0, 97, 147, 1, 0, 99, 147, 1, 0, 101, 147, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 157, 229, 1, 0, 105, 147, 1, 0, 0, 0, 0, 0, 193, 229, 1, 0, 109, 147, 1, 0, 149, 25, 2, 0, 111, 147, 1, 0, 169, 42, 2, 0, 1, 67, 2, 0, 111, 147, 1, 0, 111, 147, 1, 0, 165, 41, 2, 0, 89, 47, 2, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 205, 235, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 45, 209, 2, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 111, 147, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 53, 73, 54, 72, 10, 26, 2, 208, 7, 34, ...]]]

Flutter doctor (my project will be upgraded to null safety soon, right now i must use android chipmunk in order to compile):
[✓] Flutter (Channel stable, 2.10.5, on macOS 12.5 21G72 darwin-arm, locale en-IL)
    • Flutter version 2.10.5 at /Users/aviv/fvm/versions/2.10.5
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 5464c5bac7 (1 year, 1 month ago), 2022-04-18 09:55:37 -0700
    • Engine revision 57d3bac3dd
    • Dart version 2.16.2
    • DevTools version 2.9.2

[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.2)
    • Android SDK at /Users/aviv/Library/Android/sdk
    • Platform android-33, build-tools 33.0.2
    • Java binary at: /Applications/Android Studio 2.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.12+0-b1504.28-7817840)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 14.2)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.3

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[!] Android Studio (version 2022.2)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    ✗ Unable to find bundled Java version.
    • Try updating or re-installing Android Studio.

[✓] Android Studio (version 2021.2)
    • Android Studio at /Applications/Android Studio 2.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.12+0-b1504.28-7817840)

[✓] IntelliJ IDEA Community Edition (version 2020.3)
    • IntelliJ at /Applications/IntelliJ IDEA CE.app
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart

[✓] Connected device (2 available)
    • GM1913 (mobile) • 1190200c • android-arm64  • Android 11 (API 30)
    • Chrome (web)    • chrome   • web-javascript • Google Chrome 112.0.5615.49

[✓] HTTP Host Availability
    • All required HTTP hosts are available

I will appreciate any help.

philips77, 2023-05-29:
The firmware image file should be prepared with https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/mcuboot/imgtool.html
Signing the fw adds a MCU header to the binary with the correct firmware.

If your device isn't based on NSC or Zephyr, but nRF5 SDK, use a different library for updating the device.

AvivZem, 2023-05-29:
> The firmware image file should be prepared with https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/mcuboot/imgtool.html Signing the fw adds a MCU header to the binary with the correct [...]

Thanks, my device is in fact nRF5 based. 
does the [Native](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager) implementation work with nRF5?

philips77, 2023-05-29:
For that you need these:
- Android native: https://github.com/NordicSemiconductor/Android-DFU-Library
- iOS native: https://github.com/NordicSemiconductor/IOS-DFU-Library
- Flutter 3rd party library: https://github.com/NordicSemiconductor/Android-DFU-Library#flutter

AvivZem, 2023-05-29:
> For that you need these:
> - Android native: https://github.com/NordicSemiconductor/Android-DFU-Library
> - iOS native: https://github.com/NordicSemiconductor/IOS-DFU-Library
> - Flutter 3rd party library: https://github.com/NordicSemiconductor/Android-DFU-Library#flutter

Thanks, I will check those

closed 2023-05-29 by AvivZem