nordicsemi/Flutter-nRF-Connect-Device-Manager #28: [BUG] Inconsistent FirmwareUpgradeState `toProto` method that causes an app crash on iOS

wcoder, 2023-06-14:
Steps:
- Check on iOS
- Execute `update(...)` method
- Receives an invalid state from native library `iOSMcuManagerLibrary` in `UpdateManager.upgradeStateDidChange` method

## Android
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ba4cf9f45bba1061be6ab8e10861718fdbc8b02f/android/src/main/kotlin/no/nordicsemi/android/mcumgr_flutter/ext/FirmwareUpgradeState.kt#L7-L14

## iOS (case `.eraseAppSettings`)
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ba4cf9f45bba1061be6ab8e10861718fdbc8b02f/ios/Classes/Extensions/FirmwareUpgradeState%2BProto.swift#L13-L29

## Actual
Part of iOS device crash log:
```
Crashed: com.apple.main-thread
0  libswiftCore.dylib             0x3836c _assertionFailure(_:_:file:line:flags:) + 664
1  mcumgr_flutter                 0x5638 $s20iOSMcuManagerLibrary20FirmwareUpgradeStateO14mcumgr_flutterE7toProtoAD0j6UpdateF7ChangesVABOyF + 176
2  mcumgr_flutter                 0x19174 $s14mcumgr_flutter13UpdateManagerC21upgradeStateDidChange4from2toy06iOSMcuD7Library015FirmwareUpgradeF0O_AItF + 124
3  iOSMcuManagerLibrary           0x1f878 $s20iOSMcuManagerLibrary015FirmwareUpgradeB0C18objc_sync_setState33_C756FE8045CE31964DEB24C06DC38D0FLLyyAA0deI0OFyyScMYccfU_ + 136
4  iOSMcuManagerLibrary           0x1f8b4 $sIeg_IeyB_TR + 28
5  libdispatch.dylib              0x24b4 _dispatch_call_block_and_release + 32
6  libdispatch.dylib              0x3fdc _dispatch_client_callout + 20
7  libdispatch.dylib              0x127f4 _dispatch_main_queue_drain + 928
8  libdispatch.dylib              0x12444 _dispatch_main_queue_callback_4CF + 44
9  CoreFoundation                 0x9aa08 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
10 CoreFoundation                 0x7c368 __CFRunLoopRun + 2036
11 CoreFoundation                 0x811e4 CFRunLoopRunSpecific + 612
12 GraphicsServices               0x1368 GSEventRunModal + 164
13 UIKitCore                      0x3a2d88 -[UIApplication _run] + 888
14 UIKitCore                      0x3a29ec UIApplicationMain + 340
```

## Environment
Version: latest master

philips77, 2023-06-15:
Thank you for the report. We will look into it.

wcoder, 2023-07-18:
Potential fix in fork: https://github.com/ldoriat8/Flutter-nRF-Connect-Device-Manager/commit/974b1ded4fff7c22281a8acb2cb54934ebb693f0

jchirinosodio, 2023-09-14:
> Potential fix in fork: [ldoriat8@974b1de](https://github.com/ldoriat8/Flutter-nRF-Connect-Device-Manager/commit/974b1ded4fff7c22281a8acb2cb54934ebb693f0)

Hi there, I was able to succesfully update on my project with this hot fix, so weird I have another POC on iOS with it working, but my main project in which I tried to include my logic from the POC crashes there!

Still have not found the reason why(besides my main project being about 1 year old and my POC is like 1 month old),
Perhaps flutter has changed something for newer projects and old ones don't work out of the box?

olo95, 2023-10-05:
After implementing the fix I got next issue:

Invalid response: {"splitStatus" : 0, "images" : {{"hash" : 0xE54CC54DA7B072XE84XBDC776DB62ADXXXX2644B094317DBDB35B668AACF6208, "active" : true, "pending" : false, "confirmed" : true, "version" : "0.0.0", "slot" : 0, "bootable" : true, "permanent" : false}, {"confirmed" : false, "version" : "0.0.0", "slot" : 1, "hash" : 0xE54CC54DA7B072XE84XBDC776DB62ADXXXX2644B094317DBDB35B668AACF6208, "bootable" : true, "pending" : false, "active" : false, "permanent" : false}}}.

Do you know maybe what is this about?

philips77, 2024-01-03:
Working on it.

NickKibish, 2024-01-04:
Fixed in 0.2.0

philips77, 2024-01-04:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/e2ea5dd83240b67610ad062eeb15ca0e87b0bd15/mcumgr-core/src/main/java/io/runtime/mcumgr/dfu/task/EraseStorage.java#L26
Erasing app storage is a part of ULOAD state on Android.

NickKibish, 2024-01-05:
Version 0.2.0 has all missed cases. Now iOS and Flutter versions have the same states.