nordicsemi/Flutter-nRF-Connect-Device-Manager #33: No way to configure update mode

wcoder, 2023-07-19:
No way to configure UpdateManager mode it is always "**Confirm Only**" on both platforms.

iOS:
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ba4cf9f45bba1061be6ab8e10861718fdbc8b02f/ios/Classes/UpdateManager.swift#L39

Android:
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ba4cf9f45bba1061be6ab8e10861718fdbc8b02f/android/src/main/kotlin/no/nordicsemi/android/mcumgr_flutter/UpdateManager.kt#L49

Also, native iOS and Android libraries use TestAndConfirm mode as default mode and recommended mode.

dinesharjani, 2023-07-20:
Hi @wcoder sorry for the delay,

So for multi-image DFU, on the firmware side only .confirmOnly works due to development / software issues. It's not related to the library code itself, either the Flutter wrapper or the native Android / iOS libraries. It's because of the DFU update process itself. The library (this one, on Flutter) just tries to make it easier for you, the developers.

Let me know if you need more specific details / whatsoever.

NickKibish, 2024-01-08:
in the latest version (0.2.0) [DeviceUpdateManager](https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ff925b80d3b3b03f59b282455cd4035937a0cfed/lib/src/mcumgr_update_manager.dart#L97) accepts [FirmwareUpgradeConfiguration](https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ff925b80d3b3b03f59b282455cd4035937a0cfed/lib/src/mcumgr_flutter.dart#L28). 

So now you can set FirmwareUpgradeMode.

closed 2024-01-08 by NickKibish