nordicsemi/Flutter-nRF-Connect-Device-Manager #38: Not able to build Flutter app

rodkiewicz, 2023-10-25:
Hi
I am not able to build my flutter app.
Error build logs:
```
Swift Compiler Error (Xcode): Switch must be exhaustive
/Users/e/.pub-cache/hosted/pub.dev/mcumgr_flutter-0.1.1+1/ios/Classes/Extensions/FirmwareUpgradeState+Proto.swift:13:8

Swift Compiler Error (Xcode): Type 'McuMgrLogCategory' has no member 'config'
/Users/e/.pub-cache/hosted/pub.dev/mcumgr_flutter-0.1.1+1/ios/Classes/Extensions/McuLog+Ext.swift:27:14

Swift Compiler Error (Xcode): Cannot convert value of type '[(Int, Data)]' to expected argument type '[ImageManager.Image]'
/Users/e/.pub-cache/hosted/pub.dev/mcumgr_flutter-0.1.1+1/ios/Classes/UpdateManager.swift:40:37
``` 
Env:
  flutter: 3.13.4
  mcumgr_flutter: 0.1.1+1
Ios env:
 iOSMcuManagerLibrary (1.4)

rodkiewicz, 2023-10-25:
If i force iOSMcuManagerLibrary to version 1.3.3 in podfile, the issue does not occur.

NickKibish, 2024-01-05:
Could you try version 0.2.0. It uses iOSMcuManagerLibrary 1.4.3 and have few changes. I hope you won't face any issues in this version.

closed 2024-01-05 by NickKibish