nordicsemi/Flutter-nRF-Connect-Device-Manager #42: mcumgr_flutter is not compatible with riverpod_generator

nissaba, 2023-11-15:
If I am adding the "mcumgr_flutter: ^0.1.1+1" to my projets pubspec file that is using riverpod generator I get this error from the pub get command. 

/Users/pascale/development/flutter/bin/flutter --no-color pub get
Resolving dependencies...
    Because riverpod_generator >=2.3.3 <2.3.5 depends on riverpod_analyzer_utils ^0.3.4 and riverpod_generator >=2.3.5 <2.3.6 depends on riverpod_analyzer_utils ^0.4.1, riverpod_generator >=2.3.3 <2.3.6 requires riverpod_analyzer_utils ^0.3.4 or ^0.4.1.
    Because no versions of riverpod_generator match >2.2.5 <2.2.6 and riverpod_generator >=2.2.6 <2.3.3 depends on riverpod_analyzer_utils ^0.3.3, riverpod_generator >2.2.5 <2.3.3 requires riverpod_analyzer_utils ^0.3.3.
    Thus, riverpod_generator >2.2.5 <2.3.6 requires riverpod_analyzer_utils >=0.3.3 <0.4.0 or ^0.4.1.
    And because riverpod_generator >=2.3.6 <3.0.0-dev.2 depends on riverpod_analyzer_utils ^0.4.3, riverpod_generator >2.2.5 <3.0.0-dev.2 requires riverpod_analyzer_utils >=0.3.3 <0.4.0 or >=0.4.1 <0.5.0.
    And because riverpod_generator 2.2.5 depends on riverpod_analyzer_utils ^0.3.2 and riverpod_analyzer_utils >=0.3.3 depends on custom_lint_core ^0.5.2, riverpod_generator >=2.2.5 <3.0.0-dev.2 requires custom_lint_core ^0.5.2 or riverpod_analyzer_utils >=0.3.2 <0.3.3.
(1) So, because riverpod_analyzer_utils >=0.3.2 <0.3.3 depends on custom_lint_core ^0.5.1 and custom_lint_core >=0.5.1 <0.5.2 depends on custom_lint 0.5.1, riverpod_generator >=2.2.5 <3.0.0-dev.2 requires custom_lint 0.5.1 or custom_lint_core ^0.5.2.

    Because no versions of custom_lint_core match >0.5.2 <0.5.3 and custom_lint_core >=0.5.3 <0.5.4 depends on custom_lint 0.5.3, custom_lint_core >0.5.2 <0.5.4 requires custom_lint 0.5.3.
    And because custom_lint_core >=0.5.4 <0.5.5 depends on custom_lint 0.5.4 and custom_lint_core >=0.5.5 <0.5.6 depends on custom_lint 0.5.5, custom_lint_core >0.5.2 <0.5.6 requires custom_lint 0.5.3 or 0.5.4 or 0.5.5.
    And because riverpod_generator >=2.2.5 <3.0.0-dev.2 requires custom_lint 0.5.1 or custom_lint_core ^0.5.2 (1), riverpod_generator >=2.2.5 <3.0.0-dev.2 requires custom_lint 0.5.1 or 0.5.3 or 0.5.4 or 0.5.5 or custom_lint_core 0.5.2 or >=0.5.6 <0.6.0.
    **And because custom_lint_core 0.5.2 depends on custom_lint 0.5.2 and custom_lint_core >=0.5.6 depends on custom_lint 0.5.6, riverpod_generator >=2.2.5 <3.0.0-dev.2 requires custom_lint 0.5.1 or 0.5.2 or 0.5.3 or 0.5.4 or 0.5.5 or 0.5.6.
    And because custom_lint >=0.1.0 depends on rxdart ^0.27.7 and every version of mcumgr_flutter depends on rxdart ^0.26.0, riverpod_generator >=2.2.5 <3.0.0-dev.2 is incompatible with mcumgr_flutter.
    So, because notio depends on both mcumgr_flutter ^0.1.1+1 and riverpod_generator ^2.2.5, version solving failed.
Process finished with exit code 1**

Would be great if this problem gets fixed I am sure it is not big deal, just to update the project dependencies, hopefully not code breaking will happen.

nissaba, 2023-12-05:
all this needs if for this to use rxdart ^0.27.7 instead of rxdart ^0.26.0

philips77, 2023-12-08:
We're working on updating the library, so we'll also look at your issue. Thank you for bringing it to us.

NickKibish, 2024-01-05:
Version 0.2.0 uses rxdart 0.27.7. I hope you won't face this problem anymore.

https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ff925b80d3b3b03f59b282455cd4035937a0cfed/pubspec.lock#L133C6-L133C6

closed 2024-01-05 by NickKibish