nordicsemi/Flutter-nRF-Connect-Device-Manager #43: No way to soft reset the device after transferring image done

ptienchuan, 2023-11-20:
Hi development team,

I have no idea how to soft reset the device after transferring image done when using this package.
Please tell me:
- If you implemented this feature ?
- If you have a plan to implement in the future ?
- How can I do that now ?
Thanks!

Env:
- mcumgr_flutter: ^0.1.1+1
- iOSMcuManagerLibrary (1.3.3)

philips77, 2024-01-08:
Hello,
The protocol for updating devices using McuManager over SMP protocol is using the following commands:
* Image list (reads the current state of slots on all images)
* Image Upload (sending fw)
* Image Test and/or Confirm (sets flags that the McuBoot should swap/apply the new image)
* Reset

You may also choose a "mode" of the DFU: Test & Confirm, Test Only, Confirm Only or Upload Only. In all of them the Reset command is sent, which should reset the device. In Test & Confirm mode, after the device resets the library will reconnect to it and send Confirm command. There won't be another Reset command sent after that. Other 3 modes finish after sending Reset.

ptienchuan, 2024-01-12:
@philips77 Thanks for your response. Could you tell me how to choose "mode" ?

NickKibish, 2024-01-12:
Hi @ptienchuan,

You can set `FirmwareUpgradeConfiguration` in the `update` method in `UpdateManager.`

```dart
final updateManager = await _updateManagerFactory
        .getUpdateManager(request.peripheral!.identifier);

    updateManager.setup();
    updateManager.update(
      request.firmwareImages!,
      configuration: FirmwareUpgradeConfiguration(
        // firmwareUpgradeMode: FirmwareUpgradeMode.confirmOnly,
        // firmwareUpgradeMode: FirmwareUpgradeMode.testAndConfirm,
        // firmwareUpgradeMode: FirmwareUpgradeMode.testOnly,
        // firmwareUpgradeMode: FirmwareUpgradeMode.uploadOnly,
      ),
    );
```

closed 2024-01-14 by NickKibish