nordicsemi/Flutter-nRF-Connect-Device-Manager #47: Invalid confirm response header on IOS

EddieWilkie91, 2023-12-28:
Hi there,

I'm currently receiving this error on iOS when doing a firmware update. The upload completes but I get stuck on the confirm state and receive this error: 
```
flutter: E Invalid response: Header: {"version": "1", "op": "3", "flags": 0, "length": 244, "group": 1, "seqNum": 79, "commandId": 0}, Payload: {"images" : {{"permanent" : false, "pending" : false, "slot" : 0, "confirmed" : true, "bootable" : true, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "version" : "0.8.0", "active" : true}, {"version" : "0.8.0", "confirmed" : false, "pending" : false, "slot" : 1, "active" : false, "bootable" : true, "permanent" : false, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1}}, "splitStatus" : 0}
```

I've tested the upload using the uploadOnly mode and it works fine so seems like a problem with the confirmOnly mode. This works as expected on Android.  Is there anything that you can see wrong with this header format that would cause issues? 

Note: I understand I'm uploading the same firmware version but this still won't work with different versions.

Thanks

Setup:
mcumgr_flutter: ^0.2.0+1
flutter_blue_plus: 1.31.4 

Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.16.4, on macOS 14.0 23A344 darwin-arm64, locale en-GB)
[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 15.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2023.1)
[✓] VS Code (version 1.85.1)
[✓] Connected device (4 available)

EddieWilkie91, 2024-01-03:
Hey @philips77, thanks for the reply.

This is the full log output of the upload: 

I'm also getting an exception for an unsupported state. I think this is already logged [here](https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/issues/28)
```
flutter: Request McuMgr parameters
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Unsupported state
#0      ProtoUpdateStateToModel.convert (package:mcumgr_flutter/proto/extensions/proto_ext.dart:42:9)
#1      DeviceUpdateManager._setupUpdateStateStream.<anonymous closure> (package:mcumgr_flutter/src/mcumgr_update_manager.dart:171:37)
#2      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#3      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#4      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#5      _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#6      _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#7      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#8      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#9      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#10     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#11     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#12     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#13     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#14     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#15     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#16     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#17     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#18     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#19     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#20     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
flutter: A Upgrade started with 1 images using 'Confirm only' mode
flutter: A Device capabilities received.
flutter: D Setting SAR buffer size to 2475 bytes.
flutter: Validate
flutter: Upload
flutter: D Bootloader Mode Unknown.
flutter: I Validation response: Header: {"version": "1", "op": "1", "flags": 0, "length": 244, "group": 1, "seqNum": 212, "commandId": 0}, Payload: {"images" : {{"permanent" : false, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "slot" : 0, "version" : "0.8.0", "bootable" : true, "pending" : false, "active" : true, "confirmed" : true}, {"hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "version" : "0.8.0", "slot" : 1, "active" : false, "confirmed" : false, "pending" : false, "bootable" : true, "permanent" : false}}, "splitStatus" : 0}
flutter: 2392 / 242256} bytes sent
flutter: D Response Offset: 2392, UploadOffset: 2392, Pending Offsets: [],
flutter: D Response Offset: 4836, UploadOffset: 4836, Pending Offsets: [],
flutter: 4836 / 242256} bytes sent
flutter: 7280 / 242256} bytes sent
flutter: 9724 / 242256} bytes sent
flutter: D Response Offset: 7280, UploadOffset: 7280, Pending Offsets: [],
flutter: D Response Offset: 9724, UploadOffset: 9724, Pending Offsets: [],
flutter: 12168 / 242256} bytes sent
flutter: 14612 / 242256} bytes sent
flutter: 17056 / 242256} bytes sent
flutter: 19500 / 242256} bytes sent
flutter: D Response Offset: 12168, UploadOffset: 12168, Pending Offsets: [],
flutter: D Response Offset: 14612, UploadOffset: 14612, Pending Offsets: [],
flutter: D Response Offset: 17056, UploadOffset: 17056, Pending Offsets: [],
flutter: D Response Offset: 19500, UploadOffset: 19500, Pending Offsets: [],
flutter: 21944 / 242256} bytes sent
flutter: 24388 / 242256} bytes sent
flutter: 26832 / 242256} bytes sent
flutter: D Response Offset: 21944, UploadOffset: 21944, Pending Offsets: [],
flutter: D Response Offset: 24388, UploadOffset: 24388, Pending Offsets: [],
flutter: D Response Offset: 26832, UploadOffset: 26832, Pending Offsets: [],
flutter: 29276 / 242256} bytes sent
flutter: 31720 / 242256} bytes sent
flutter: 34164 / 242256} bytes sent
flutter: D Response Offset: 29276, UploadOffset: 29276, Pending Offsets: [],
flutter: D Response Offset: 31720, UploadOffset: 31720, Pending Offsets: [],
flutter: D Response Offset: 34164, UploadOffset: 34164, Pending Offsets: [],
flutter: D Response Offset: 36608, UploadOffset: 36608, Pending Offsets: [],
flutter: 36608 / 242256} bytes sent
flutter: 39052 / 242256} bytes sent
flutter: 41496 / 242256} bytes sent
flutter: 43940 / 242256} bytes sent
flutter: D Response Offset: 39052, UploadOffset: 39052, Pending Offsets: [],
[... 151 lines cut ...]
flutter: 227240 / 242256} bytes sent
flutter: 229684 / 242256} bytes sent
flutter: D Response Offset: 224796, UploadOffset: 224796, Pending Offsets: [],
flutter: D Response Offset: 227240, UploadOffset: 227240, Pending Offsets: [],
flutter: D Response Offset: 229684, UploadOffset: 229684, Pending Offsets: [],
flutter: 232128 / 242256} bytes sent
flutter: 234572 / 242256} bytes sent
flutter: 237016 / 242256} bytes sent
flutter: D Response Offset: 232128, UploadOffset: 232128, Pending Offsets: [],
flutter: D Response Offset: 234572, UploadOffset: 234572, Pending Offsets: [],
flutter: D Response Offset: 237016, UploadOffset: 237016, Pending Offsets: [],
flutter: 239460 / 242256} bytes sent
[... cut ...]
flutter: E Invalid response: Header: {"version": "1", "op": "3", "flags": 0, "length": 244, "group": 1, "seqNum": 57, "commandId": 0}, Payload: {"images" : {{"hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "permanent" : false, "pending" : false, "slot" : 0, "bootable" : true, "version" : "0.8.0", "active" : true, "confirmed" : true}, {"hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "confirmed" : false, "slot" : 1, "bootable" : true, "permanent" : false, "pending" : false, "active" : false, "version" : "0.8.0"}}, "splitStatus" : 0}
```

dinesharjani, 2024-01-08:
This looks to me like the firmware is sending CONFIRM command to Image 0 Slot 1, but after sending the command, the response is "confirmed: false", hence why an error is thrown. This explains why in upload only mode, there's no error since it doesn't send the CONFIRM command.

philips77, 2024-01-08:
In the snippet above, both slots contain the same image. Sending Confirm would switch `pending` and `permanent` flags, but as the image is already `active`, it does nothing. But this is expected.

Does it work for you when you upload a different image?

EddieWilkie91, 2024-01-08:
@philips77 It doesn't work with a different image (I mentioned this in the original post). I tried with a newer build, this is the log after the upload progress reaches 100%: 

Note the upload works using confirmOnly mode using the nRFConnect app

```
flutter: Erase app settings
flutter: D Response Offset: 239460, UploadOffset: 239460, Pending Offsets: [],
flutter: D Response Offset: 241904, UploadOffset: 241904, Pending Offsets: [],
flutter: D Response Offset: 241972, UploadOffset: 241972, Pending Offsets: [],
flutter: A Upload finished (1 of 1)
flutter: Confirm
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Invalid response: Header: {"version": "1", "op": "3", "flags": 0, "length": 244, "group": 1, "seqNum": 136, "commandId": 0}, Payload: {"images" : {{"active" : true, "version" : "0.8.0", "pending" : false, "slot" : 0, "permanent" : false, "bootable" : true, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "confirmed" : true}, {"bootable" : true, "slot" : 1, "permanent" : true, "confirmed" : false, "hash" : 0xD198797CE849C86DCD996DC4F19815FD170E1223AD9AC39D05D22EBDB6ED62F5, "pending" : true, "active" : false, "version" : "0.8.1"}}, "splitStatus" : 0}
flutter: A Erasing app settings completed
flutter: I Confirmation response: Header: {"version": "1", "op": "3", "flags": 0, "length": 244, "group": 1, "seqNum": 136, "commandId": 0}, Payload: {"images" : {{"active" : true, "version" : "0.8.0", "pending" : false, "slot" : 0, "permanent" : false, "bootable" : true, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "confirmed" : true}, {"bootable" : true, "slot" : 1, "permanent" : true, "confirmed" : false, "hash" : 0xD198797CE849C86DCD996DC4F19815FD170E1223AD9AC39D05D22EBDB6ED62F5, "pending" : true, "active" : false, "version" : "0.8.1"}}, "splitStatus" : 0}
flutter: E Invalid response: Header: {"version": "1", "op": "3", "flags": 0, "length": 244, "group": 1, "seqNum": 136, "commandId": 0}, Payload: {"images" : {{"active" : true, "version" : "0.8.0", "pending" : false, "slot" : 0, "permanent" : false, "bootable" : true, "hash" : 0xEA964E848A79122C45C19502460028ACC7414D1095EBE7C868C3D892ED4978E1, "confirmed" : true}, {"bootable" : true, "slot" : 1, "permanent" : true, "confirmed" : false, "hash" : 0xD198797CE849C86DCD996DC4F19815FD170E1223AD9AC39D05D22EBDB6ED62F5, "pending" : true, "active" : false, "version" : "0.8.1"}}, "splitStatus" : 0}
```

dinesharjani, 2024-01-08:
So it works using nRF Connect app on iOS, but fails with the iOS Flutter library?

philips77, 2024-01-09:
@dinesharjani in the last log, the new image got successfully set the *pending* and *permanent* flags:
```json
{
    "images": [
        {
            "active": true,
            "version": "0.8.0",
            "pending": false,
            "slot": 0,
            "permanent": false,
            "bootable": true,
            "hash": 106106775537712004222562768737963993252138343382193496563092467289099573229793,
            "confirmed": true
        },
        {
            "bootable": true,
            "slot": 1,
            "permanent": true, <--
            "confirmed": false,
            "hash": 94802784586553142175149626055645777376980395540838902018887105462859109065461,
            "pending": true, <--
            "active": false,
            "version": "0.8.1"
        }
    ],
    "splitStatus": 0
}
```
Any idea why that causes Invalid Response?

philips77, 2024-01-09:
@NickKibish Flutter lib is using iOS nRF Connect Device Manager in version 1.4.3, right?
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/ff925b80d3b3b03f59b282455cd4035937a0cfed/ios/mcumgr_flutter.podspec#L18

@EddieWilkie91 could you change the version to 1.5.3 and try?

EddieWilkie91, 2024-01-09:
Hey @philips77,

I tested this using '1.5.3', I had to make some updates to the UpdateManager.swift as the 'hash' property was required in the start function. But this gave the same error. 

The following made me double check my code: 

> Note lack of "imageIndex" param, but that's OK. It defaults to 0.

I set the imageIndex to '1' as I wasn't sure from the flutter example documentation what this value was for: 

```
// `firmware` is a List of data and index pairs
final List<Tuple2<int, Uint8List>> firmwareScheme = ...;
final configuration = const FirmwareUpgradeConfiguration(
      estimatedSwapTime: const Duration(seconds: 0),
      byteAlignment: ImageUploadAlignment.fourByte,
      eraseAppSettings: true,
      pipelineDepth: 1,
    );
// `configuration` is an optional parameter. If not provided, default values will be used.
updateManager.update(firmware.firmwareScheme, configuration: configuration);
```
Changing this to '0' no longer gives me the error and the upload seems to be successful now. Sorry for this, it was my lack of understanding on multi image uploads. I also think the reason that it worked on Android never led me to check this. 

Thanks for all the help in debugging this.

philips77, 2024-01-09:
Image Index is the core number. Main/single core is always 0, secondary is 1, etc.

closed 2024-01-09 by philips77