nordicsemi/Flutter-nRF-Connect-Device-Manager #66: Feature request: method to read image list (or at least active image hash)

rstieger, 2024-03-15:
I would like a method in the FirmwareUpdateManager to get the current image list (or at least the hashes).

For devices that are not bonded but use random MAC addresses, the testAndConfirm mode won't work (because the MAC address changes, it can't automatically reconnect). Currently I can run testOnly, verify the firmware version string (from DIS), and then run confirmOnly. However, it would be more reliable if I could verify the image hash instead of the version string.

I understand that allowing firmware updates without bonding is not best practice, but there are some use cases where it is necessary. I can also imagine other use cases where being able to read the image list - to show what images are already on a device, before starting an update - would be beneficial.

savvn001, 2024-03-15:
Would like this too. Implementation of the mcumgr image list feature.

NickKibish, 2024-03-29:
Hi, @rstieger 

Great news! I [implemented](https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/3fd1260f36aa694536500686f86643700c92b57d/lib/src/mcumgr_update_manager.dart#L210) this feature in 0.4.0 🎉

Example:
```dart
final images = await _firmwareUpdateManager!.readImageList();
if (images != null) {
  images.forEach((image) {
    print("Image: ${image.image}");
    print("Slot ${image.slot}");
    print("Version: ${image.version}");
    print("Hash: ${image.hashString}");
    print("Bootable: ${image.bootable}");
    print("Pending: ${image.pending}");
    print("Confirmed: ${image.confirmed}");
    print("Active: ${image.active}");
    print("Permanent: ${image.permanent}");
  });
}
```

I will close this issue if it's what you've expected.

closed 2024-03-29 by NickKibish