nordicsemi/Flutter-nRF-Connect-Device-Manager #83: HELP: new firmware not loaded after update

nissaba, 2024-04-23:
Hi, I must be missing a step. When I upload a firmware it will do all the validation, then upload, and finally call reset. 
when the ble device comes back it is still using the old firmware. 
if I re-upload the firmware I then see in the log that sending only takes 1 like and says 100/100, So by this I know the new firmware is in memory but was not moved to take the place of the old one. 

I tested the firmware via the example app, this does work. I tried to look in the code of the example but the code is so scattered I couldn't find where the upload code was or most importantly were the termination reset code was.

philips77, 2024-04-24:
> When I upload a firmware, it will do all the validation, then upload, and finally call reset.

It seems like you're not doing the _confirm_ or _test_ step. What mode are you using for the DFU?
You may set the mode in `FirmwareUpgradeConfiguration.firmwareUpgradeMode`.
I recommend `CONFIRM_ONLY`, as this will upload, confirm and reset, after which the device should boot with the new firmware. As far as I know, the default upload mode is `TEST_ONLY` which may not work on some devices, like nRF53, which cannot "test" the net-core.

You may encounter the following cases:
1. After upload the new fw is in secondary slot but wasn't moved to the primary slot.
   * This can be verified using nRF Connect Device Manager apps on Android and iOS
   * That means it has not been confirmed/tested. Check the _upload mode_ and logs. After _confirm_ state the image in the secondary slot should have `pending` and `permanent` flags set.
2. After update and reboot the new firmware is gone (secondary slot is empty).
   * That means, that the fw was signed with a wrong key and signatures didn't match on verification - McuBoot had removed the image.
   * I guess this isn't your case, as second upload is immediate - image is still in the secondary slot.
3. The image is in the secondary slot, but it has `active` flag set
   * That means, you're using Direct XIP feature allowing to boot an image from any of the slots to skip swapping.

nissaba, 2024-04-29:
Thanks. The info was helpfull into finding the problem. It was onthe firmware side, the firmware engineer told me that the file containing the firmware versiona nd build was not saved and so the new firmware kept the old version info and so the device was not swtiching to the new one, or in any case we could not see the deffirence when reading the version info characteristic.

closed 2024-04-29 by nissaba