nordicsemi/Flutter-nRF-Connect-Device-Manager #86: When project imports mcumgr_flutter, iOS requests for Bluetooth permissions at startup

olo95, 2024-05-13:
Should it be requested when methods of the plugin are in use?

NickKibish, 2024-05-13:
You're absolutely correct.

The Bluetooth permissions issue you've raised was indeed addressed following a request in #81, and it's been resolved in version [0.4.1](https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/releases/tag/0.4.1). 

To see the fix in action, you can refer to this [demo project](https://github.com/NickKibish/McuMgr-Simplest-Demo/blob/2f73dbbc4395633b42eecb68263a93794f45ceb8/lib/main.dart#L38). Let me know if you need further assistance!

olo95, 2024-05-13:
Sorry, completely forgot to upgrade, 0.4.1 changed approach to my problem. Thank you for quick response :)

closed 2024-05-13 by olo95