nordicsemi/Flutter-nRF-Connect-Device-Manager #88: A problem occurred configuring project ':mcumgr_flutter'

anhvu2503, 2024-06-24:
When I add dependency with mcumgr_flutter version: ^0.4.1
It was showing error like this with android
[image]

NickKibish, 2024-06-24:
Try to run the project from Android Studio to get a more specific error message.

In my case, it was an incompatible Kotlin version. 
[image]

anhvu2503, 2024-06-25:
This is my bug
[image]

closed 2024-06-25 by NickKibish

anhvu2503, 2024-12-03:
@MichaelQun 
You should download the library locally, place it in a specific folder, and reference the path to the folder where the library is stored.
This will resolve the issue.