nordicsemi/Flutter-nRF-Connect-Device-Manager #90: “The handle is invalid“ error thrown right at the end of firmware upgrade

Wes1324, 2025-01-14:
I work for a major purchaser of Nordic products and we have a critical issue in an app that we are launching this month.

If iOS users pair and bond with our bluetooth device and then perform a firmware upgrade without restarting the app, an error is thrown saying "The handle is invalid" (see following screenshot). The error does not occur if users restart the app between bonding with the device and upgrading its firmware, however this is not a feasible solution to our problem as it would massively disrupt the user experience of our app.

[image]

I have performed the firmware upgrade using the example app of this package with the same `FirmwareUpgradeMode` and `estimatedSwapTime` that our app uses (`testAndConfirm` and 90 seconds respectively) and I experience the same error (see following screenshot).

[image]
[image]

Is this a known issue? Do you have guidance on how to stop this error being thrown please?

Thanks in advance for any help you can provide.

philips77, 2025-01-14:
Hello,

Did you try with `uploadOnly` mode, in which library doesn't reconnect to confirm, but does it immediately after upload? 
Do you set `eraseAppStorage` (I don't remember the exact param name) to false, so that bond data aren't removed? I think yes, as I don't see it in the logs.
Did you measure how long it takes the device to boot again after the update to set the estimate time to 90 sec?
Did you try a different phone and iOS version?

philips77, 2025-01-15:
I'm talking about this one:
https://github.com/NordicSemiconductor/Flutter-nRF-Connect-Device-Manager/blob/a33bffdd8da82164446deeea0f3744b6892853a0/lib/src/mcumgr_flutter.dart#L9
Check if you have it set to `false`, otherwise it will erase app storage, including bond information.

Also, does your update modifies the list of services? This "invalid handle" error comes from the peripheral when central is trying an operation on non-existing handle. If it does, peripheral should notify cental using Service Change indication, which would trigger a new service discovery.

Wes1324, 2025-01-15:
Thanks for your replies @philips77 

> Did you try with uploadOnly mode, in which library doesn't reconnect to confirm, but does it immediately after upload?

I did not try uploadOnly mode, mainly because most of the docs seem to discourage using this mode. I did however try the `confirmOnly` mode, which doesn't cause the "handle is invalid" but the app doesn't seem to be connected to the device anymore after using this mode.

Is it known that the device will be disconnected after performing an upgrade in the uploadOnly mode? If so, I assume the intention is that we manually reconnect to the device after a performing a firmware upgrade in this mode?

> Do you set eraseAppStorage (I don't remember the exact param name) to false, so that bond data aren't removed? I think yes, as I don't see it in the logs.

Yes, we do set this to false

> Did you measure how long it takes the device to boot again after the update to set the estimate time to 90 sec?

Yes, we measured it a few times and it takes roughly 70-75 seconds, so we decided to up it to 90 seconds to be sure the swap would have completed by the time the reboot occurs

> Did you try a different phone and iOS version?

A few different people have encountered this issue so the likelihood is that it's occurred on different types of iPhone and iOS versions. I couldn't tell you exactly which versions right now but I could get that information if it would be helpful.

> Also, does your update modifies the list of services? This "invalid handle" error comes from the peripheral when central is trying an operation on non-existing handle. If it does, peripheral should notify cental using Service Change indication, which would trigger a new service discovery.

I'm not sure about this. I can ask our firmware developers tomorrow. Is this "Service Change indication" something that the bluetooth device should send to the phone? And then that will trigger the phone to do a new service discovery?

Thanks again

philips77, 2025-01-16:
> I did however try the confirmOnly mode, which doesn't cause the "handle is invalid" but the app doesn't seem to be connected to the device anymore after using this mode.

Sorry, this is what I meant. Indeed, in `confirmOnly` the lib is sending Reset command and reports success, so you need to reconnect on your own.

philips77, 2025-01-16:
> Is this "Service Change indication" something that the bluetooth device should send to the phone? And then that will trigger the phone to do a new service discovery?

Yes. If a peripheral changes the attribute table (i.e. on firmware update, or by changing mode) it is supposed to notify the client by sending a Service Change indication. The indications should be enabled automatically on bonding, and this is what iOS does. Upon receiving the notification central invalidates cached services and triggers new service discovery. You should receive [peripheral(_:didModifyServices:)](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/peripheral(_:didmodifyservices:)) callback from where you may call `discoverServices()` again.

This indication can also be sent when the device changed services on restart. Then the indication is sent immediately after reconnection and establishing encryption.

But first check if the device indeed changed services. You may try to connect to it before and after the update using Android nRF Connect and selecting "Refresh services" from the menu. Compare the services in both cases.

Wes1324, 2025-01-28:
Thanks for your replies.

> first check if the device indeed changed services.

I did this check and one service is removed in the newer firmware version

> it is supposed to notify the client by sending a Service Change indication

I checked with our firmware devs and it seems that everything is in place to send the indication. I added the code outlined in the flutter_blue_plus documentation [here](https://pub.dev/packages/flutter_blue_plus#services-changed-characteristic). Unfortunately, the code inside the callback is never invoked so it seems the indication isn't being received by the app for some reason.

I have added a fix in the meantime, which does work but feels a bit hacky - disconnecting and reconnecting the device before starting the firmware upgrade.

I still don't understand why the hack mentioned above or restarting the app before performing the upgrade stops the "handle is invalid" error from being thrown. The services are still changing once the upgrade is complete so I would expect the error to be thrown despite having disconnected the device or restarted the app before doing the upgrade...

philips77, 2025-01-29:
Android incorrectly does service discovery on each connection, even when bonded. We reported it to Google so it may change in the future. So when you reconnect, the service (for now) get invalidated and you get new ones - hance the handle numbers are now correct.

Bluetooth spec says, that on bonded devices the service discovery should be done once after bonding and then may be retriggered using Service Change indication of Database Hash characteristic. Android in some versions never caches, in some versions always caches (even for non-bonded with Service Changed characteristic, where it shouldn't) so I recommend trying on different Android phones and versions. Some time ago they added `onServiceChange` callback, which can retrigger serivce discovery. We tried it and it gets called on Android 14 when SC indication is received.

iOS, from our experience, is behaving correctly.

Wes1324, 2025-01-30:
That's really useful to know. Thanks for that info.

I have one more question please. I can see that our firmware exposes a Service Changed characteristic under the "Generic Attribute" service.

[image]

However, when I try to access that "Generic Attribute" service from inside the app code (see code below), I get null.

[image]

Should the app be able to access this service? I'm wondering if the fact that the app can't access this service is the reason why the app isn't reacting to the Service Changed indications from that characteristic, resulting in the "handle is invalid" error.

philips77, 2025-02-04:
Hi,
This service isn't acessible on iOS. Perhaps your framework tries to emulate the same behavior on Android as well, for example thrying to managing it by itself?

dinesharjani, 2026-08-19:
Hi @Wes1324 - I take it we can mark this issue as resolved?

Wes1324, 2026-08-19:
Yes, thank you for your help with this issue. I will close it now

closed 2026-08-19 by Wes1324