nordicsemi/Flutter-nRF-Connect-Device-Manager #92: Flutter app BLE device search & send characteristic

nomanhamid, 2025-01-24:
I am trying to add BLE device search and send data from the phone to the device in Dart - Flutter, but unable to achieve the objective.
Earlier, I had implemented the same process in Kotlin (Native Android App) via [NRF Blinky Example](https://github.com/NordicSemiconductor/Android-nRF-Blinky).

Please clarify if my direction is wrong OR if any other example/lib is available so I can follow the same process in my Dart App.

The process is as follows:

1. Search for BLE devices
2. Send characteristic to the BLE device via phone (in my case FobID)
3. If everything goes successful then the Nordic BLE device clicks.

I really appreciate any help you can provide.

philips77, 2025-01-24:
In Flutter there's https://pub.dev/packages/flutter_blue_plus which you may use to scan and connect to BLE devices.
1. scan
2. select one
3. connect
4. service discovery -> list of services and characteristics
5. write a value to any characteritic

Committee-A, 2025-01-24:
> In Flutter there's https://pub.dev/packages/flutter_blue_plus which you may use to scan and connect to BLE devices.
> 
> 1. scan
> 2. select one
> 3. connect
> 4. service discovery -> list of services and characteristics
> 5. write a value to any characteritic

Thanks @philips77 .
Let me try this one.

Committee-A, 2025-03-25:
Hi @philips77 ,

I have attempted multiple approaches to transmit data to the device, but all have been unsuccessful. However, I was able to successfully discover, list, and establish a connection with BLE devices.

An additional issue I am encountering is that once a BLE device is connected, it automatically disconnects after some time, even without any data being written. Furthermore, whenever I attempt to reconnect and write data (byteArray) to the BLE device, it consistently disconnects.

I would appreciate any guidance on what I might be doing wrong, as I am using the default write function for this purpose. The only remaining challenge is writing data to the device, for which I am seeking assistance.

FYI: I'm following the same flutter package you suggested in the above reply.