nordicsemi/Flutter-nRF-Connect-Device-Manager #96: [Android MinSDK] Why flutter package's android minsdk version is higher than "Android-nRF-Connect-Device-Manager"?

cogito666, 2025-04-08:
Hello there,

While I am checking README for below two repo, a question came up on me.

Why flutter package's android minsdk version is higher than "Android-nRF-Connect-Device-Manager"?

Is there a particular reason? 🤔🤔🤔🤔🤔

[image]

[image]

philips77, 2025-04-09:
Hello,
Although they both don't use any specific API for Android 4.4 or 5, the feature to support longet MTU than 23 on GATT layer was added in API 21 (Android 5). For Android 4.3 and 4.4 you need to limit the size of your SMP packets, so many features won't work unless you enable reassembly on the device side.

closed 2025-04-09 by philips77

cogito666, 2025-04-09:
> Hello, Although they both don't use any specific API for Android 4.4 or 5, the feature to support longet MTU than 23 on GATT layer was added in API 21 (Android 5). For Android 4.3 and 4.4 you need [...]

I understood
> minsdk = 21 is recommeded